luftdaten-at/firmware #35: send data to sensor.community

silvioheinze, 2024-10-14:
Beschreibung: https://github.com/opendata-stuttgart/meta/wiki/APIs

mittels Einstellungsmöglichkeit in den settings.toml ob die Daten gesendet werden.

n11ik, 2024-10-30:
Ich habe gesehen, dass die namen der Dimensionen anders sind als bei uns und dass es nicht alle Dimensionen gibt die wir messen.
Daher habe ich mal die unterstützten Dimensions IDs in dem dictionary übersetzt:

```
PM0_1: "P01",
PM1_0: "P1",
PM2_5: "P2",
PM4_0: "P4",
PM10_0: "P10",
HUMIDITY: "humidity",
TEMPERATURE: "temperature",
PRESSURE: "pressure",
CO2: "co2_ppm",
O3: "ozone_ppb",
TVOC: "tvoc",
NO2: "no2_ppb",
```
Ich glaube das sollten alle sein.

Hast du dir das so vorgestellt?

n11ik, 2024-10-30:
@silvioheinze

Was ist bei: "X-Sensor: esp8266-12345678" mit eindeutiger id gemeint?
Was soll bei: "your_version" angegeben werden? 
```
// header
Content-Type: application/json  
X-Pin: 7  
X-Sensor: esp8266-12345678
// data
{
"software_version": "your_version", 
    "sensordatavalues":[
        {"value_type":"temperature","value":"22.30"},
        {"value_type":"humidity","value":"34.70"}
    ]
}
```

closed 2024-11-04 by n11ik