luftdaten-at/firmware #49: AirStation crashes after ~1h

n11ik, 2024-11-12:
The crash is due to the following exception:
```
GET faild: Error connecting socket: [Errno 119] EINPROGRESS, 
first error: [Errno 116] ETIMEDOUT
2024-11-12T09:52:50.000Z [DEBUG] process file: json_queue/tmp_log.txt
Traceback (most recent call last):
  File "adafruit_connection_manager.py", line 240, in _get_connected_socket
OSError: [Errno 119] EINPROGRESS

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "main.py", line 310, in <module>
  File "main.py", line 303, in main
  File "models/air_station.py", line 307, in tick
  File "models/air_station.py", line 243, in send_to_api
  File "wifi_client.py", line 88, in send_json_to_api
  File "adafruit_requests.py", line 525, in request
  File "adafruit_connection_manager.py", line 339, in get_socket
RuntimeError: Error connecting socket: [Errno 119] EINPROGRESS
```

silvioheinze, 2024-11-12:
@n11ik es Unterschiede zwischen Circuitpython Versionen? Insbesondere 9.1 und 9.2?

n11ik, 2024-11-13:
Auf dem circuitpython github gibt es ein Open issue das ähnlich ist: https://github.com/adafruit/circuitpython/issues/9302

n11ik, 2024-11-14:
Ich habe über nacht 3 AirStations, zwei mit der 9.1 und eine mit der 9.2, laufen lassen und der Fehler ist bis jetzt noch nicht wieder aufgetreten.
Warscheinlich ist der Fehler entstanden, weil kurzzeitig keine Internet verbindung vorhanden war.
Daher werde ich das Issue mal schlißen.

closed 2024-11-14 by n11ik